rt-net/RaspberryPiMouse #15: build_install.ubuntu14.bash error

neo0117, 2019-04-21:
make clean
make -C /usr/linux-headers-`uname -r` M=`pwd` V=1 clean
make[1]: Entering directory '/usr/src/linux-headers-4.4.38-v7+'
make[1]: ***No rule to make target 'clean'. Stop.
make[1]: Leaving directory '/usr/src/linux-headers-4.4.38-v7+'
Makefile:10: recipe for target 'clean' failed
make: *** [clean] Error 2

ShotaAk, 2019-04-22:
Please write more details about your system (e.g OS, Kernel Version).

If you use Raspbian, you have to run **build_install.raspbian.bash** instead of build_install.ubuntu14.bash.

ShotaAk, 2019-04-22:
The installation is updated with #17.
Could try [new installation](https://github.com/rt-net/RaspberryPiMouse/blob/53218abf5ef2fa8c43771a0339794b35d211f4f5/README.md) and  tell us the result?

neo0117, 2019-04-22:
Hi ShotaAk,
Thank you very much for the quick response. I am using Ubuntu16 and ran build_install.ubuntu14.bash.
I am currently on a business trip and will have access to my PiMouse later on Thursday. 

I will first try the updated installation script and let you know.

Thank you very much for your help :)

neo0117, 2019-04-26:
Hi,
I tried the new installation... The results are:
sudo apt install linux-headers-$(uname -r)

Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package linux-headers-4.4.38-v7
E: Couldn't find any package by glob 'linux-headers-4.4.38-v7'
E: Couldn't find any package by regex 'linux-headers-4.4.38-v7'

My OS: ubuntuMATE 16.04
uname -r => 4.4.38-v7+

Should I change to a different OS?

Thank you

Tiryoh, 2019-04-26:
Hi @neo0117,
On Ubuntu MATE, you can try installing kernel header files by running:
```
sudo apt install raspberrypi-kernel-headers
```

I wrote an entry here (written in Japanese):
https://memoteki.net/archives/1086

https://translate.google.com/translate?hl=en&sl=auto&tl=en&u=https%3A%2F%2Fmemoteki.net%2Farchives%2F1086

neo0117, 2019-04-26:
Hi @Tiryoh 
I just tried it... and the result is:

Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package raspberrypi-kernal-headers

So it seems it is not in the repository... What should I do next?

Thank you

Tiryoh, 2019-04-26:
> E: Unable to locate package raspberrypi-kernal-headers

It appears that the package name is wrong.
Trying to install raspberrypi-*kernel*-headers might work.

neo0117, 2019-04-26:
Hi @Tiryoh 
I used the old fashion way in your link to download the "deb" package from phil.lavin.me.uk and install the kernel headers successfully. Thank you very much!

But when I excute the "./build_install.bash",  it shows "Segmentation fault" 
sudo insmod rtmouse.ko ---> Segmentation fault

By the way, I am using RaspberryPi 3 Model B+. If this is the problem?

Tiryoh, 2019-04-26:
Hmm..., I don't use Raspberry Pi 3 Model B+, but that sounds strange...
Could you show me the results of these 3 commands?

```
uname -a
lsb_release -a
modinfo rtmouse.ko
```

neo0117, 2019-04-26:
Hi,
After executing the "./build_install.bash", I got the following ERROR messages:

ERROR: Kernel configuration is invalid;
               include /generated/autoconf.h or include /config.auto.conf are missing;
               Run 'make oldconfig && make prepare' on kernel src to fix it.

So how do I "Run 'make oldconfig && make prepare' "?

neo0117, 2019-04-26:
1. uname -a: 
Linux jdlee-rpi 4.4.38-v7+ #938 SMP Thu Dec 15 15:22:21 GMT 2016 armv7l armv7l armv7l GNU/Linux

2.  lsb_release -a
No LSB modules are available.
Distribution ID: Ubuntu
Description:      Ubuntu 16.04.6 LTS
Codename:       xenial

3. modinfo rtmouse.ko
filename:     /home/jdlee/RaspberryPiMouse/src/drivers/rtmouse.ko
description: Raspberry Pi Mouse device driver
version: 1:1.4
license: GPL
author: RT Corporation
srcversion: 3BDEC1C95E8255A4B777C62
alias: spi:mcp3204
alias: i2c:rtcounter_r
alias: i2c:rtcounter_l
depends:
vermagic: 4.4.38-v7 SMP mod_unload modversions ARMv7

Tiryoh, 2019-04-30:
Hi @neo0117,

> So how do I "Run 'make oldconfig && make prepare' "?

I think you can run the commands in `/usr/src/linux-$(uname -r)`.
I cannot say for sure, but I guess that the Debian package you download from `phil.lavin.me.uk` didn't installed successfully.
Therefore, I recommend the following method.
1. uninstall the linux kernel headers which you installed by Debian package
```sh
sudo dpkg -r linux-headers-4.4.38-v7+_4.4.38-v7+-2_armhf #Since I can not confirm the package name, please supplement by yourself.
```
2. install the linux kernel headers with `apt` command (the `apt` command will support installing the kernel header files)
```sh
sudo apt install raspberrypi-kernel-headers
```
3. try again to install the device driver
```sh
./build_install.bash
```

neo0117, 2019-05-01:
Hi,
I tried your approach, but still have no luck :(
I then tried both Raspbian and NOOBS OS and installed the "raspberrypi-kernel-headers". Then when I: ./build_install.bash

The errors showed up:

echo >&2 "  ERROR: Kernel configuration is invalid.";		\
echo >&2 "         include/generated/autoconf.h or include/config/auto.conf are missing.";\
echo >&2 "         Run 'make oldconfig && make prepare' on kernel src to fix it.";	\
..............
..............

+ sudo insmod rtmouse.ko
/home/pi/RaspberryPiMouse/utils/build_install.ubuntu14.bash: line 10:  1888 Segmentation fault      sudo insmod rtmouse.ko

Can you tell me which OS image file did you use and if there any place I can download it?

Thank you very much for your help.

neo0117, 2019-05-01:
Hi,
I also tried the prebuild image file from https://downloads.ubiquityrobotics.com/.
Everytime when I run "sudo apt install raspberrypi-kernel-headers", it always shows:
raspberrypi-kernel-headers is already the newest version 

But when I run ./build_install.bash, it says:

++ dirname /home/ubuntu/RaspberryPiMouse/utils/build_install.ubuntu14.bash
+ dir=/home/ubuntu/RaspberryPiMouse/utils/../
[ ! -e /usr/src/linux-headers-$(uname -r) ] && { bash -e $dir/utils/print_env.bash "No kernel header files found."; exit 1; }
uname -r
++ uname -r
+ '[' '!' -e /usr/src/linux-headers-4.9.80-v7+ ']'
+ bash -e /home/ubuntu/RaspberryPiMouse/utils/..//utils/print_env.bash 'No kernel header files found.'
===================
ERROR: No kernel header files found.

It seems to me that all problems are related to the kernels files....

Tiryoh, 2019-05-02:
> I tried your approach, but still have no luck :(

That's too bad...

> The errors showed up:
>
> echo >&2 " ERROR: Kernel configuration is invalid.";	
> echo >&2 " include/generated/autoconf.h or include/config/auto.conf are missing.";
> echo >&2 " Run 'make oldconfig && make prepare' on kernel src to fix it.";	

I guess that is not an error, its just a debug output written in `/usr/src/linux-headers-$(uname -r)/Makefile`.  Is that log like this?
```
$ make
make -C /usr/src/linux-headers-`uname -r` M=`pwd` V=1 modules
make[1]: Entering directory '/usr/src/linux-headers-4.4.38-v7+'
test -e include/generated/autoconf.h -a -e include/config/auto.conf || (		\
echo >&2;							\
echo >&2 "  ERROR: Kernel configuration is invalid.";		\
echo >&2 "         include/generated/autoconf.h or include/config/auto.conf are missing.";\
echo >&2 "         Run 'make oldconfig && make prepare' on kernel src to fix it.";	\
echo >&2 ;							\
/bin/false)
mkdir -p /home/ubuntu/RaspberryPiMouse/src/drivers/.tmp_versions ; rm -f /home/ubuntu/RaspberryPiMouse/src/drivers/.tmp_versions/*
make -f ./scripts/Makefile.build obj=/home/ubuntu/RaspberryPiMouse/src/drivers
  gcc -Wp,-MD,/home/ubuntu/RaspberryPiMouse/src/drivers/.rtmouse.o.d  -nostdinc -isystem /usr/lib/gcc/arm-linux-gnueabihf/4.7/include -I./arch/arm/include -Iarch/arm/include/generated/uapi -Iarch/arm/include/generated  -Iinclude -I./arch/arm/include/uapi -Iarch/arm/include/generated/uapi -I./include/uapi -Iinclude/generated/uapi -include ./include/linux/kconfig.h -D__KERNEL__ -mlittle-endian -Iarch/arm/mach-bcm2709/include -Wall -Wundef -Wstrict-prototypes -Wno-trigraphs -fno-strict-aliasing -fno-common -Werror-implicit-function-declaration -Wno-format-security -std=gnu89 -fno-PIE -fno-dwarf2-cfi-asm -fno-omit-frame-pointer -mapcs -mno-sched-prolog -fno-ipa-sra -mabi=aapcs-linux -mno-thumb-interwork -mfpu=vfp -funwind-tables -marm -D__LINUX_ARM_ARCH__=7 -march=armv7-a -msoft-float -Uarm -fno-delete-null-pointer-checks -Wno-maybe-uninitialized -O2 --param=allow-store-data-races=0 -Wframe-larger-than=1024 -fno-stack-protector -Wno-unused-but-set-variable -fno-omit-frame-pointer -fno-optimize-sibling-calls -fno-var-tracking-assignments -pg -Wdeclaration-after-statement -Wno-pointer-sign -fno-strict-overflow -fconserve-stack -Werror=implicit-int -Werror=strict-prototypes -DCC_HAVE_ASM_GOTO  -DMODULE  -D"KBUILD_STR(s)=#s" -D"KBUILD_BASENAME=KBUILD_STR(rtmouse)"  -D"KBUILD_MODNAME=KBUILD_STR(rtmouse)" -c -o /home/ubuntu/RaspberryPiMouse/src/drivers/.tmp_rtmouse.o /home/ubuntu/RaspberryPiMouse/src/drivers/rtmouse.c
  if [ "-pg" = "-pg" ]; then if [ /home/ubuntu/RaspberryPiMouse/src/drivers/rtmouse.o != "scripts/mod/empty.o" ]; then ./scripts/recordmcount  "/home/ubuntu/RaspberryPiMouse/src/drivers/rtmouse.o"; fi; fi;
(cat /dev/null;   echo kernel//home/ubuntu/RaspberryPiMouse/src/drivers/rtmouse.ko;) > /home/ubuntu/RaspberryPiMouse/src/drivers/modules.order
make -f ./scripts/Makefile.modpost
  find /home/ubuntu/RaspberryPiMouse/src/drivers/.tmp_versions -name '*.mod' | xargs -r grep -h '\.ko$' | sort -u | sed 's/\.ko$/.o/' | scripts/mod/modpost -m -a -i ./Module.symvers -I /home/ubuntu/RaspberryPiMouse/src/drivers/Module.symvers  -o /home/ubuntu/RaspberryPiMouse/src/drivers/Module.symvers -S  -w  -s -T -
  gcc -Wp,-MD,/home/ubuntu/RaspberryPiMouse/src/drivers/.rtmouse.mod.o.d  -nostdinc -isystem /usr/lib/gcc/arm-linux-gnueabihf/4.7/include -I./arch/arm/include -Iarch/arm/include/generated/uapi -Iarch/arm/include/generated  -Iinclude -I./arch/arm/include/uapi -Iarch/arm/include/generated/uapi -I./include/uapi -Iinclude/generated/uapi -include ./include/linux/kconfig.h -D__KERNEL__ -mlittle-endian -Iarch/arm/mach-bcm2709/include -Wall -Wundef -Wstrict-prototypes -Wno-trigraphs -fno-strict-aliasing -fno-common -Werror-implicit-function-declaration -Wno-format-security -std=gnu89 -fno-PIE -fno-dwarf2-cfi-asm -fno-omit-frame-pointer -mapcs -mno-sched-prolog -fno-ipa-sra -mabi=aapcs-linux -mno-thumb-interwork -mfpu=vfp -funwind-tables -marm -D__LINUX_ARM_ARCH__=7 -march=armv7-a -msoft-float -Uarm -fno-delete-null-pointer-checks -Wno-maybe-uninitialized -O2 --param=allow-store-data-races=0 -Wframe-larger-than=1024 -fno-stack-protector -Wno-unused-but-set-variable -fno-omit-frame-pointer -fno-optimize-sibling-calls -fno-var-tracking-assignments -pg -Wdeclaration-after-statement -Wno-pointer-sign -fno-strict-overflow -fconserve-stack -Werror=implicit-int -Werror=strict-prototypes -DCC_HAVE_ASM_GOTO  -D"KBUILD_STR(s)=#s" -D"KBUILD_BASENAME=KBUILD_STR(rtmouse.mod)"  -D"KBUILD_MODNAME=KBUILD_STR(rtmouse)" -DMODULE  -c -o /home/ubuntu/RaspberryPiMouse/src/drivers/rtmouse.mod.o /home/ubuntu/RaspberryPiMouse/src/drivers/rtmouse.mod.c
  ld -EL -r  -T ./scripts/module-common.lds --build-id  -o /home/ubuntu/RaspberryPiMouse/src/drivers/rtmouse.ko /home/ubuntu/RaspberryPiMouse/src/drivers/rtmouse.o /home/ubuntu/RaspberryPiMouse/src/drivers/rtmouse.mod.o
make[1]: Leaving directory '/usr/src/linux-headers-4.4.38-v7+'
```

> sudo insmod rtmouse.ko
> /home/pi/RaspberryPiMouse/utils/build_install.ubuntu14.bash: line 10: 1888 Segmentation fault sudo insmod rtmouse.ko

Can you show me the full log?  I'm not sure, but I guess SPI and I2C are not enabled.
https://www.raspberrypi.org/documentation/configuration/raspi-config.md

> Can you tell me which OS image file did you use and if there any place I can download it?

[... cut ...]
What does the script show?  The script `./build_install.bash` outputs its environment information which is necessary for debugging, for example like this: https://github.com/rt-net/RaspberryPiMouse/pull/17#issue-272275824
```
===================
ERROR: No kernel header files found.
If you need someone's support, you should share this information.
Linux raspberrypi 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 9.8 (stretch)
Release:	9.8
Codename:	stretch
ls: cannot access '/usr/src/linux*': No such file or directory
===================
```

neo0117, 2019-05-02:
Hi,
I think it might actually work this time!

I (1) ignore the debugger messages and (2) enable the SPI and I2C. The outputs are now:

sudo insmod rtmouse.ko
+ sudo insmod rtmouse.ko
sleep 1
+ sleep 1
sudo chmod 666 /dev/rt*
+ sudo chmod 666 /dev/rtbuzzer0 /dev/rtled0 /dev/rtled1 /dev/rtled2 /dev/rtled3 /dev/rtlightsensor0 /dev/rtmotor0 /dev/rtmotoren0 /dev/rtmotor_raw_l0 /dev/rtmotor_raw_r0 /dev/rtswitch0 /dev/rtswitch1 /dev/rtswitch2
echo 0 > /dev/rtmotoren0
+ echo 0

When I ls /dev/rt*.... I can see the following in the /dev directory.

/dev/rtbuzzer0  /dev/rtled3          /dev/rtmotor_raw_l0  /dev/rtswitch2
/dev/rtled0     /dev/rtlightsensor0  /dev/rtmotor_raw_r0
/dev/rtled1     /dev/rtmotor0        /dev/rtswitch0
/dev/rtled2     /dev/rtmotoren0      /dev/rtswitch1

Does this mean the drivers are successfully installed?

Tiryoh, 2019-05-02:
> When I ls /dev/rt*.... I can see the following in the /dev directory.
>
> /dev/rtbuzzer0 /dev/rtled3 /dev/rtmotor_raw_l0 /dev/rtswitch2
> /dev/rtled0 /dev/rtlightsensor0 /dev/rtmotor_raw_r0
> /dev/rtled1 [...]

Sounds good.  If you don't see any errors in `dmesg` log, it means okay.
By the way, which OS did you tried on?

neo0117, 2019-05-02:
Where is the dmesg log located?

You mentioned that it could be the SPI and I2C not being enabled earlier and I knew that in Raspbian we need to enable it manually. So I tried Rasbian today.

By the way, I would also like to try Ubuntu. From the link [https://wiki.ubuntu.com/ARM/RaspberryPi](url), there is ARMHF and ARM64, which one should I download?

Thanks a lot!

Tiryoh, 2019-05-03:
`dmesg` log is an output of `dmesg` command.  There is a lot of information about dmesg, you can google it ;)
http://man7.org/linux/man-pages/man1/dmesg.1.html

> there is ARMHF and ARM64, which one should I download?

Either is good, but armhf has more information on the internet than arm64.

neo0117, 2019-05-05:
I ran this command: dmesg --level=err,warn

It did not show any errors, but with the following lines:\

brcmfmac: brcmf_fw_map_chip_to_name: using brcm/brcmfmac43430-sdio.bin for chip 0x00a9a6(43430) rev 0x000001
brcmfmac: brcmf_c_preinit_dcmds: Firmware version = wl0: Oct 23 2017 03:55:53 version 7.45.98.38 (r674442 CY) FWID 01-e58d219f
brcmfmac: brcmf_c_preinit_dcmds: CLM version = API: 12.2 Data: 7.11.15 Compiler: 1.24.2 ClmImport: 1.24.1 Creation: 2014-05-26 10:53:55 Inc Data: 9.10.39 Inc Compiler: 1.29.4 Inc ClmImport: 1.36.3 Creation: 2017-10-23 03:47:14 

Should I be worried?

Thank you very much!

Tiryoh, 2019-05-08:
> Should I be worried?

It seems like an issue of the WiFi module.  I'm not familiar about WiFi module, so if you're concerned, please visit the Raspberry Pi forum.
https://www.raspberrypi.org/forums/viewtopic.php?t=218445

closed 2019-08-29 by ShotaAk