rt-net/RaspberryPiMouse #48: rtcounterが見当たらない

shu-rt, 2020-06-02:
# 不具合の概要

build_install.bashをした際に、rtcounterが入らない。

# 実行環境

- OS: Raspbian Buster (2020-02-13-raspbian-buster-full.img)
- ソフトのバージョン: 2020/06/02のmasterブランチ (コミット af0c46af71488b02ebf0451384462092ac3ddc2e)

# 再現方法

1. 2020-02-13-raspbian-buster-full.imgが入ったmicroSDカードを使ってラズパイマウスを起動
2. 設定でSSH, I2C, SPIなどをENABLEにして再起動。
3. RaspberryPiMouseをgit cloneする。
4. raspberrypi-kernel-headersをinstallする。
5. build_install.bashを実行する。

# 期待する動作

/dev/にrtcounterが存在して欲しい

# ログ・画像

dmesgを実行すると、以下のメッセージが表示される。
```sh
[   65.005554] rtmouse: rtcounter not found, or wrong i2c device probed
[   65.006389] rtmouse: rtcounter not found, or wrong i2c device probed
```

# その他

```sh
sudo apt install i2c-tools
i2cdetect -y 1
```
を実行したところ
```sh
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: 10 11 -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --
```
という結果をえた。

```sh
i2cget -y 1 0x10 0x10 w | sed 's/x\(..\)\(..\)/x\2\1/g'
```
を実行すると
```sh
0x1010
```
という結果を得た。

```sh
sudo i2cset -y 1 0x10 0x10 0x00 0x00 I

```
を実行すると
```sh
0x0000
```
という結果を得た。

また、`uname -a`の実行結果は以下の通りである。
```
Linux raspberrypi 4.19.97-v7+ #1294 SMP Thu Jan 30 13:15:58 GMT 2020 armv7l GNU/Linux
```

Tiryoh, 2020-06-04:
I2Cのボーレートの設定が必要な可能性があります。
関連 #51 #13

shu-rt, 2020-06-08:
`dtparam=i2c_baudrate=62500`を`/boot/config.txt`に書き込んで
再起動してから`build_install.bash`を行ったら無事解決しました、ありがとうございました。

closed 2020-06-08 by shu-rt