rt-net/RaspberryPiMouse #52: makeコマンドが見つらないのでbuild-essentialのインストールをREADMEに追加して欲しい

ShotaAk, 2020-06-05:
# 現状の問題点

Ubuntu 18.04をインストールした後、./utils/build_install.bashを実行しましたが下記のエラーが出ます。

```txt
ubuntu@ubuntu:~/RaspberryPiMouse/utils$ ./build_install.bash 
#!/bin/bash -vxe

dir=$(dirname $0)/../
++ dirname /home/ubuntu/RaspberryPiMouse/utils/build_install.ubuntu14.bash
+ dir=/home/ubuntu/RaspberryPiMouse/utils/../
[ ! -e /usr/src/linux-headers-$(uname -r) ] && { bash -e $dir/utils/print_env.bash "No kernel header files found."; exit 1; }
++ uname -r
+ '[' '!' -e /usr/src/linux-headers-5.3.0-1026-raspi2 ']'
cd $dir/src/drivers/
+ cd /home/ubuntu/RaspberryPiMouse/utils/..//src/drivers/
rm Makefile
+ rm Makefile
ln -s Makefile.ubuntu14 Makefile
+ ln -s Makefile.ubuntu14 Makefile
make clean
+ make clean
/home/ubuntu/RaspberryPiMouse/utils/build_install.ubuntu14.bash: line 8: make: command not found
```

# 改善案

makeがインストールされていないので、`sudo apt install build-essential`をREADMEのインストール手順に追加して欲しい。

Tiryoh, 2020-06-05:
@ShotaAk ちなみにこれはどこに公開されているイメージですか？

ShotaAk, 2020-06-05:
ここです。
https://ubuntu.com/download/raspberry-pi

closed 2020-06-05 by Tiryoh