rt-net/RaspberryPiMouse #58: コントリビュータの皆様へ：GPLv2からGPLv3への変更について

ShotaAk, 2020-06-30:
本リポジトリはライセンスを**GPLv3**として公開していますが、**GPLv2**で公開されているコード及びライブラリを使用しているため、ライセンス違反とならないようにGPLv2へ変更します。

そこで、GPLv3が適用される前提でコントリビュートして頂いた皆様にお願いです。

- ライセンス変更に異論がなければ:+1:でリアクションをお願いします。
- ライセンス変更に意見がある場合はコメントをお願いします。

皆様の合意を得た上でライセンスを変更します。

--- 

コントリビュータ一覧（直近のPRを抜粋しています）：

@Tiryoh 
https://github.com/rt-net/RaspberryPiMouse/pull/51

@shu-rt 
https://github.com/rt-net/RaspberryPiMouse/pull/50

@AtsushiSaito 
https://github.com/rt-net/RaspberryPiMouse/pull/12

@tabe 
https://github.com/rt-net/RaspberryPiMouse/pull/5

@ryuichiueda 
https://github.com/rt-net/RaspberryPiMouse/pull/2

---

- ソースコードの流用：
  -  [take-iwiw/DeviceDriverLesson](https://github.com/take-iwiw/DeviceDriverLesson)
    - **GPL/BSD License**
  - [mcp3204.c in Raspberry Piで学ぶARMデバイスドライバープログラミング](http://www.socym.co.jp/support/s-940#ttlDownload)
    - **GPL v2 License**
  - [RPi-Distro/raspi-gpio](https://github.com/RPi-Distro/raspi-gpio)
    - The 3-Clause BSD License
- ライブラリの使用：
  - https://github.com/rt-net/RaspberryPiMouse/blob/c28a0656d85b7650ff831446a6a672661b980782/src/drivers/rtmouse.c#L25-L47

Tiryoh, 2020-06-30:
2020年6月30日18時39分（JST）現在 未確認ユーザ

- @aoki-rt
- @shu-rt

Tiryoh, 2020-07-01:
2020年7月1日10時35分（JST）、7ユーザ全員のリアクションを確認しました。

ShotaAk, 2020-07-01:
リアクションありがとうございました。
#57 でライセンスを変更します。

closed 2020-07-01 by Tiryoh