rt-net/RaspberryPiMouse #67: デバドラで5.10.11-v7+の対応予定はありますか？

ForexRobotics, 2021-03-03:
# 現状の問題点
現在取扱説明書_2020_03_01.PDFに従い、「５．デバイスドライバーのインストールについて」を実施しようとしています。
uname -r でバージョンを確認すると 5.10.11-7+ と返ってきます。
しかしgithub上のlib/Pi2B+以下に該当フォルダのrtmouse.koは存在しないのでinsmodができない状態です。

# 改善案
5.10.11-v7+用のrtmouse.koの提供をお願いしたいです。

# 代替案
ソースから自分でビルドできるのであれば手順を教示いただけると助かります。

# その他
よろしくお願いいたします。

ShotaAk, 2021-03-04:
カーネルバージョン 5.10でソースビルドできるように対応中です。

https://github.com/rt-net/RaspberryPiMouse/pull/66 にてビルドテスト環境を構築しており、その対応後にソースファイルを修正する予定です。
また、このPRにはバージョン5.10でのビルドエラーについても記述しています。

ソースビルドの手順は[README](https://github.com/rt-net/RaspberryPiMouse/blob/master/README.md) に記載しています。

ForexRobotics, 2021-03-05:
情報有難うございます！内容確認いたしました。

ShotaAk, 2021-04-16:
@ForexRobotics #69 にてKernel 5.10に対応致しました。ご確認ください。

closed 2021-04-21 by ShotaAk