rt-net/RaspberryPiMouse #73: can`t access rtbuzzer0 , rtmotor_raw_l0,r0  , rtcounter_l1,l0

NizarMhatli, 2021-08-31:
os : ubuntu 20.04 lts 
raspberry pi 4 model b 
already set up dtparam=i2c_baudrate=62500

i can access the leds , lightsensors and the switch but can access the motor, buzzer or the counter 

if i try to use cat /dev/rtcounter_l0 i will get 
no such file or directory 
if i try to echo /dev/rtmotor_raw_l0
i get  permission denied 
i don't know from where is this issue i followed the install instructions as described in this page please tell me if i missed anything

NizarMhatli, 2021-08-31:
after i run dmesg output: 
[  155.794088] rtmouse: loading out-of-tree module taints kernel.
[  155.795376] rtmouse: loading driver...
[  155.796458] rtmouse: rtcounter not found, or wrong i2c device probed
[  155.797418] rtmouse: rtcounter not found, or wrong i2c device probed
[  155.907556] rtmouse: mcp3204 probed
[  155.907684] rtmouse: 17 devices loaded.
[  155.907689] rtmouse: module installed at 4294931273
[  374.754148] motor write

NizarMhatli, 2021-08-31:
sudo i2cdetect -y 1
output : 
 0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                
printf "%d\n" 0x$(xxd -ps /sys/class/i2c-adapter/i2c-1/of_node/clock-frequency)
62500

NizarMhatli, 2021-08-31:
ubuntu@ubuntu:~$ echo 0 ~ 20000 > /dev/rtbuzzer0
-bash: echo: write error: Invalid argument
ubuntu@ubuntu:~$ cat /dev/rtbuzzer0
cat: /dev/rtbuzzer0: Invalid argument

NizarMhatli, 2021-08-31:
ubuntu@ubuntu:~$ echo 0 ~ ±10000 > /dev/rtmotor_raw_l0
-bash: echo: write error: Invalid argument
ubuntu@ubuntu:~$ echo 0 ~ ±10000 > /dev/rtmotor_raw_r0
-bash: echo: write error: Invalid argument

Tiryoh, 2021-08-31:
Hi @NizarMhatli, thanks for your feedback.

## counter
The counter device file (`rtcounter_*_0`) requires microcontroller for pulse counting.
Since microcontroller for pulse counting does not exist on Raspberry Pi Mouse V2, please consider using the option.
https://www.rt-shop.jp/index.php?main_page=product_info&cPath=1299_1395&products_id=3767

## motor
"0 ~ ±10000" means that the device file can accept values from -10000 to 10000.
For example, try the following command.
```
echo 400 > /dev/rtmotor_raw_r0
```

## buzzer
"0 ~ 20000" means that the device file can accept values from 0 to 20000.
For example, try the following command.
```
echo 400 > /dev/rtbuzzer0
```

NizarMhatli, 2021-08-31:
thank you for your response i tried running these commands the work on the terminal but no action on the robot the motors nor the buzzer change state , the buzzer works i can hear it but it doesn't change state/sound output when i use this command  echo 400 > /dev/rtbuzzer0, the motor wont also start the robot is powered on im using a 11 v battery i hope it not hardware issue ...

Tiryoh, 2021-09-01:
Did you tried the sample program? I hope it helps.
https://github.com/rt-net/RaspberryPiMouse/tree/master/SampleProgram

> the buzzer works i can hear it but it doesn't change state/sound output when i use this command echo 400 > /dev/rtbuzzer0

The command `echo 400 > /dev/rtbuzzer0` only sound the buzzer at 400Hz. If you want to sound it in other frequency, chagning the value `400` to other will be able to change the sound.

> the motor wont also start the robot is powered on im using a 11 v battery i hope it not hardware issue ...

11.0V is a little low, but I guess this will not be a problem.
There are three steps to get the motor running, please make sure you are doing them well one at a time.

1. turn on the hardware motor SW
2. turn on the software motor SW
3. command the frequency

[image]

NizarMhatli, 2021-09-01:
thank you for your response , tried different echo commands for the buzzer nothing changes , followed your instructions but still the motors wont start  it either the battery is inefficient or the driver add on is not working properly ...

Tiryoh, 2021-09-02:
> tried different echo commands for the buzzer nothing changes , followed your instructions but still the motors wont start it either the battery is inefficient or the driver add on is not working properly ...

If it doesn't work with the sample program, there is a possibility of a malfunction.
Since I cannot determine whether it is a problem or not with the information I have now, please contact support with the actual command you entered and a video of it running.
The contact information can be found on page 48 of the manual below. The email address of the support is not written here to prevent spam.
https://rt-net.jp/wp-content/uploads/2020/04/raspberryPiMouse%E5%8F%96%E6%89%B1%E3%81%84%E8%AA%AC%E6%98%8E%E6%9B%B8_2020_03_01.pdf

> another question for the power switches can i leave the software switch off and power the raspi using its power supply will this create problems ?

It is not likely to cause a failure even if you operate the motor with the hardware SW and software SW both or either turned off. Of course, if the switch is off, the motor will not run.

Is the robot unable to use the LEDs or button switches while the motor SW is turned on? If so, the power supply to the robot may be unstable.

Tiryoh, 2021-09-02:
>  (raspi powered by charger)

Let me confirm, does this means that supplying the power from the robot,  not from the microUSB/USB Type-C port of Raspberry Pi?

> software switch off /motor switch on leds/switch/sensor work fine and the test step codes for the run fine (raspi powered by charger)
> using the 11v battery only motor switch on/software switch on , raspi doesn't boot up

Battery power may be insufficient. Please check the battery status. If it has been used for a long time and has deteriorated, it may not work.
The following batteries have been confirmed.
* Battery type: Li-Po 3C 11.1V
* Capacity: 1000mAh

NizarMhatli, 2021-09-02:
1- its powered from the type c port of the raspberry pie but the leds/switch/sensor work so there isn't an issue ? (are there any risks of doing so) 
2- i will test a new battery

Tiryoh, 2021-09-02:
> its powered from the type c port of the raspberry pie but the leds/switch/sensor work so there isn't an issue ? (are there any risks of doing so)

It is not recommended to supply power from Type-C port on Raspberry Pi to robot.
Some of the Type-C power supplies have a higher voltage and may damage the DCDC converter of the robot itself.
The motor is not driven by 5V, so you need to supply power directly to the robot to run the motor.

Tiryoh, 2021-10-05:
Closing this issue due to no update.
Please feel free to re-open if you need anything!

closed 2021-10-05 by Tiryoh