rt-net/RaspberryPiMouse #77: Support Kernel > 5.12

Tiryoh, 2022-06-27:
# 不具合の概要

5.12より新しいバージョンのカーネルではspi_transferに"delay_usecs" fieldがないため、`rtmouse.c`をそのままビルドしようとするとエラーが出ます

# 実行環境

- OS: Raspberry Pi OS (Raspbian GNU/Linux 11 (bullseye))
- ソフトのバージョン: https://github.com/rt-net/RaspberryPiMouse/tree/53066c2c41ba766da563b10c1a7751c0a5d60d4a

# 再現方法

1. `./utils/build_install.bash ` を実行する
4. 不具合が発生する

# 期待する動作

エラーが出ずにinsmodまで完了する

# ログ・画像

```
$ ./utils/build_install.bash 
build_install_header_from_apt_raspi2.bash
make -C /usr/src/linux-headers-5.15.32-v7+ M=/home/pi/RaspberryPiMouse/src/drivers V=0 clean
make[1]: Entering directory '/usr/src/linux-headers-5.15.32-v7+'
make[1]: Leaving directory '/usr/src/linux-headers-5.15.32-v7+'
make -C /usr/src/linux-headers-5.15.32-v7+ M=/home/pi/RaspberryPiMouse/src/drivers V=0 modules
make[1]: Entering directory '/usr/src/linux-headers-5.15.32-v7+'
  CC [M]  /home/pi/RaspberryPiMouse/src/drivers/rtmouse.o
/home/pi/RaspberryPiMouse/src/drivers/rtmouse.c: In function ‘mcp3204_probe’:
/home/pi/RaspberryPiMouse/src/drivers/rtmouse.c:1720:12: error: ‘struct spi_transfer’ has no member named ‘delay_usecs’
 1720 |  data->xfer.delay_usecs = 0;
      |            ^
make[2]: *** [scripts/Makefile.build:277: /home/pi/RaspberryPiMouse/src/drivers/rtmouse.o] Error 1
make[1]: *** [Makefile:1868: /home/pi/RaspberryPiMouse/src/drivers] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-5.15.32-v7+'
make: *** [Makefile:9: rtmouse.ko] Error 2

$ uname -a
Linux raspberrypi 5.15.32-v7+ #1538 SMP Thu Mar 31 19:38:48 BST 2022 armv7l GNU/Linux
pi@raspberrypi:~/RaspberryPiMouse $ lsb
lsblk        lsb_release  

$ lsb_release -a
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 11 (bullseye)
Release:	11
Codename:	bullseye
```

# その他

このコミットで取り除かれている
https://github.com/raspberrypi/linux/commit/3ab1cce553378fc0df1b1d26d7e23d03bd4dd3b6

delay.valueを使うようになっている
https://github.com/raspberrypi/linux/commit/05d8a019eb057d14cdf9483318a7ee8b35a69cda

delay.valueはこのコミットで追加されたもよう
https://github.com/raspberrypi/linux/commit/bebcfd272df648542c458d28fbd6a8f9428b5310

delay_usecは0で初期化してるので取り除いても問題ない可能性がある
https://github.com/rt-net/RaspberryPiMouse/blob/53066c2c41ba766da563b10c1a7751c0a5d60d4a/src/drivers/rtmouse.c#L1720

ryuichiueda, 2022-10-12:
いまこの変更↓で動作確認しようとしています。spiなので影響が出るのはブザーとモータでしょうか？

```
-       data->xfer.delay_usecs = 0;
+       data->xfer.delay.value = 0;
```

Tiryoh, 2022-10-17:
@ryuichiueda 確認遅くなりました。
SPIはライトセンサ（/dev/rtlightsensor0）が影響を受け、ブザーとモータはSPIの影響を直接は受けません。
提案の変更でrtlightsensor0の値が取得できるようであれば、さしあたり問題ないと考えます。masterブランチの内容については社内で確認が取れてから更新する予定です（提案の方法とは別の方法で更新する可能性があります）

ryuichiueda, 2022-10-17:
> ブザーとモータはSPIの影響を直接は受けません。

ありゃ、そうでしたか。失礼しました。おそらくハードウェアの問題でひとつ距離センサの値がとれなかったので、もう一度検証します。

ryuichiueda, 2022-10-19:
こちら機体を取り替えて確認しました。センサの値、全部取れました。大丈夫です。

Tiryoh, 2022-10-21:
フィードバックありがとうございます！

PR #78 (2023-06-26): Support Kernel > 5.12

# What does this implement/fix?

https://github.com/rt-net/RaspberryPiMouse/issues/77 を修正します。

Raspberry Pi OS Buster（2019年リリース）は初期段階ではKernel 5.4だったので、Kernel 5.4〜5.15でビルドできるように変更をしています。

本pull request内では11ファイルの変更が発生していますが、`rtmouse.c`ファイル以外の変更はビルドテストにのみ関連しています。

* `rtmosue.c`：Kernel 5.12以降でビルドできるようにするための変更
* その他：ビルドテストの変更

より古いカーネルの4.xについてはビルドチェックを停止しています。

# Does this close any currently open issues?

close #77

# How has this been tested?

HW: Raspberry Pi Mouse V3 + Raspberry Pi 4 Model B

* 5.15.76-v7l+, Raspbian GNU/Linux 11 (bullseye)
* 5.10.17-v7l+, Raspbian GNU/Li [... cut]
closed 2023-06-26 by ShotaAk