rt-net/RaspberryPiMouse #81: Ubuntu 24.04 Server でデバイスドライバのビルドに失敗する

KuraZuzu, 2024-07-08:
# 不具合の概要

Ubuntu24.04 Server の環境でデバイスドライバのビルドをすると失敗する。

# 実行環境

- Ubuntu 24.04 Server (kernel: 6.8.0-1006-raspi): 
- 2024年7月8日のmasterの最新（commit: 4f0ebea） 

# 再現方法

[RT Corporation Software Tutorials -> Raspberrry Pi Mouse ->デバイスドライバのインストール](https://rt-net.github.io/tutorials/raspimouse/driver/install.html)の手順に従う。ただし、OSは`Ubuntu 24.04`を使用する。

1. Raspberry Pi のOSとして`Ubuntuの24.04 Server`のイメージを選択したmicroSDカードを用意する
2. インストール手順に従う（パッケージの更新、ソフトウェアのダウンロード・インストール等）
7. `$ ./build_install.bash`を実行してビルド・インストールをする
8. ビルドの途中で不具合が発生する

# 期待する動作

ビルドに成功する

# ログ・画像

ビルド時に以下のエラーが出力される

```bash
$ ./build_install.bash
build_install_header_from_apt_raspi4.bash
make -C /usr/src/linux-headers-6.8.0-1006-raspi M=/home/ubuntu/RaspberryPiMouse/src/drivers V=0 clean
make[1]: Entering directory '/usr/src/linux-headers-6.8.0-1006-raspi'
make[1]: Leaving directory '/usr/src/linux-headers-6.8.0-1006-raspi'
make -C /usr/src/linux-headers-6.8.0-1006-raspi M=/home/ubuntu/RaspberryPiMouse/src/drivers V=0 modules
make[1]: Entering directory '/usr/src/linux-headers-6.8.0-1006-raspi'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: aarch64-linux-gnu-gcc-13 (Ubuntu 13.2.0-23ubuntu4) 13.2.0
  You are using:           gcc-13 (Ubuntu 13.2.0-23ubuntu4) 13.2.0
  CC [M]  /home/ubuntu/RaspberryPiMouse/src/drivers/rtmouse.o
/home/ubuntu/RaspberryPiMouse/src/drivers/rtmouse.c:323:15: error: initialization of ‘void (*)(struct spi_device *)’ from incompatible pointer type ‘int (*)(struct spi_device *)’ [-Werror=incompatible-pointer-types]
  323 |     .remove = mcp3204_remove,
      |               ^~~~~~~~~~~~~~
/home/ubuntu/RaspberryPiMouse/src/drivers/rtmouse.c:323:15: note: (near initialization for ‘mcp3204_driver.remove’)
/home/ubuntu/RaspberryPiMouse/src/drivers/rtmouse.c:360:14: error: initialization of ‘int (*)(struct i2c_client *)’ from incompatible pointer type ‘int (*)(struct i2c_client *, const struct i2c_device_id *)’ [-Werror=incompatible-pointer-types]
  360 |     .probe = rtcnt_i2c_probe,
      |              ^~~~~~~~~~~~~~~
/home/ubuntu/RaspberryPiMouse/src/drivers/rtmouse.c:360:14: note: (near initialization for ‘i2c_counter_driver.probe’)
/home/ubuntu/RaspberryPiMouse/src/drivers/rtmouse.c:361:15: error: initialization of ‘void (*)(struct i2c_client *)’ from incompatible pointer type ‘int (*)(struct i2c_client *)’ [-Werror=incompatible-pointer-types]
  361 |     .remove = rtcnt_i2c_remove,
      |               ^~~~~~~~~~~~~~~~
/home/ubuntu/RaspberryPiMouse/src/drivers/rtmouse.c:361:15: note: (near initialization for ‘i2c_counter_driver.remove’)
/home/ubuntu/RaspberryPiMouse/src/drivers/rtmouse.c:1120:6: warning: no previous prototype for ‘update_signed_count’ [-Wmissing-prototypes]
 1120 | void update_signed_count(struct rtcnt_device_info *dev_info, int rtcnt_count)
      |      ^~~~~~~~~~~~~~~~~~~
/home/ubuntu/RaspberryPiMouse/src/drivers/rtmouse.c:1156:6: warning: no previous prototype for ‘reset_signed_count’ [-Wmissing-prototypes]
 1156 | void reset_signed_count(struct rtcnt_device_info *dev_info, int rtcnt_count)
      |      ^~~~~~~~~~~~~~~~~~
In file included from ./arch/arm64/include/asm/alternative.h:9,
                 from ./arch/arm64/include/asm/lse.h:14,
                 from ./arch/arm64/include/asm/cmpxchg.h:14,
                 from ./arch/arm64/include/asm/atomic.h:16,
                 from ./include/linux/atomic.h:7,
                 from ./include/asm-generic/bitops/atomic.h:5,
                 from ./arch/arm64/include/asm/bitops.h:25,
                 from ./include/linux/bitops.h:68,
                 from ./include/linux/kernel.h:23,
                 from ./include/linux/cpumask.h:10,
                 from ./include/linux/smp.h:13,
                 from ./include/linux/lockdep.h:14,
                 from ./include/linux/mutex.h:17,
                 from ./include/linux/kernfs.h:11,
[... 193 lines cut ...]
  228 | struct class * __must_check class_create(const char *name);
      |                                          ~~~~~~~~~~~~^~~~
/home/ubuntu/RaspberryPiMouse/src/drivers/rtmouse.c:1932:34: error: too many arguments to function ‘class_create’
 1932 |         dev_info->device_class = class_create(THIS_MODULE, DEVNAME_CNTL);
      |                                  ^~~~~~~~~~~~
./include/linux/device/class.h:228:29: note: declared here
  228 | struct class * __must_check class_create(const char *name);
      |                             ^~~~~~~~~~~~
/home/ubuntu/RaspberryPiMouse/src/drivers/rtmouse.c: At top level:
/home/ubuntu/RaspberryPiMouse/src/drivers/rtmouse.c:2097:5: warning: no previous prototype for ‘dev_init_module’ [-Wmissing-prototypes]
 2097 | int dev_init_module(void)
      |     ^~~~~~~~~~~~~~~
/home/ubuntu/RaspberryPiMouse/src/drivers/rtmouse.c:2254:6: warning: no previous prototype for ‘dev_cleanup_module’ [-Wmissing-prototypes]
 2254 | void dev_cleanup_module(void)
      |      ^~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[3]: *** [scripts/Makefile.build:243: /home/ubuntu/RaspberryPiMouse/src/drivers/rtmouse.o] Error 1
make[2]: *** [/usr/src/linux-headers-6.8.0-1006-raspi/Makefile:1926: /home/ubuntu/RaspberryPiMouse/src/drivers] Error 2
make[1]: *** [Makefile:240: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.8.0-1006-raspi'
make: *** [Makefile:9: rtmouse.ko] Error 2

```

# その他

ShotaAk, 2024-07-10:
## class createのエラーについて

```sh
/home/ubuntu/RaspberryPiMouse/src/drivers/rtmouse.c:1322:21: error: too many arguments to function ‘class_create’
 1322 |         class_led = class_create(THIS_MODULE, DEVNAME_LED);
```

引数からownerへのポインタが削除されていました。

https://docs.kernel.org/driver-api/infrastructure.html#c.class_create

KuraZuzu, 2024-07-16:
## 修正概要

関数`class_create`のエラーに対応してビルドしました。

commit: 9a3044f

## 手順

関数`class_create`をkernel6.4.0以降と以前で処理を切り替えるように修正してビルドを実施しました。

例：
https://github.com/rt-net/RaspberryPiMouse/blob/9a3044f3b796b21026cbe593290297116190e463/src/drivers/rtmouse.c#L1322

## 結果

関数`class_create`に関するエラーは解消されました。依然として、他のエラーは残っています。

```bash
~/RaspberryPiMouse/utils$ ./build_install.bash 
build_install_header_from_apt_raspi4.bash
make -C /usr/src/linux-headers-6.8.0-1006-raspi M=/home/ubuntu/RaspberryPiMouse/src/drivers V=0 clean
make[1]: Entering directory '/usr/src/linux-headers-6.8.0-1006-raspi'
make[1]: Leaving directory '/usr/src/linux-headers-6.8.0-1006-raspi'
make -C /usr/src/linux-headers-6.8.0-1006-raspi M=/home/ubuntu/RaspberryPiMouse/src/drivers V=0 modules
make[1]: Entering directory '/usr/src/linux-headers-6.8.0-1006-raspi'
warning: the compiler differs from the one used to build the kernel
  The kernel was built by: aarch64-linux-gnu-gcc-13 (Ubuntu 13.2.0-23ubuntu4) 13.2.0
  You are using:           gcc-13 (Ubuntu 13.2.0-23ubuntu4) 13.2.0
  CC [M]  /home/ubuntu/RaspberryPiMouse/src/drivers/rtmouse.o
/home/ubuntu/RaspberryPiMouse/src/drivers/rtmouse.c:323:15: error: initialization of ‘void (*)(struct spi_device *)’ from incompatible pointer type ‘int (*)(struct spi_device *)’ [-Werror=incompatible-pointer-types]
  323 |     .remove = mcp3204_remove,
      |               ^~~~~~~~~~~~~~
/home/ubuntu/RaspberryPiMouse/src/drivers/rtmouse.c:323:15: note: (near initialization for ‘mcp3204_driver.remove’)
/home/ubuntu/RaspberryPiMouse/src/drivers/rtmouse.c:360:14: error: initialization of ‘int (*)(struct i2c_client *)’ from incompatible pointer type ‘int (*)(struct i2c_client *, const struct i2c_device_id *)’ [-Werror=incompatible-pointer-types]
  360 |     .probe = rtcnt_i2c_probe,
      |              ^~~~~~~~~~~~~~~
/home/ubuntu/RaspberryPiMouse/src/drivers/rtmouse.c:360:14: note: (near initialization for ‘i2c_counter_driver.probe’)
/home/ubuntu/RaspberryPiMouse/src/drivers/rtmouse.c:361:15: error: initialization of ‘void (*)(struct i2c_client *)’ from incompatible pointer type ‘int (*)(struct i2c_client *)’ [-Werror=incompatible-pointer-types]
  361 |     .remove = rtcnt_i2c_remove,
      |               ^~~~~~~~~~~~~~~~
/home/ubuntu/RaspberryPiMouse/src/drivers/rtmouse.c:361:15: note: (near initialization for ‘i2c_counter_driver.remove’)
/home/ubuntu/RaspberryPiMouse/src/drivers/rtmouse.c:1120:6: warning: no previous prototype for ‘update_signed_count’ [-Wmissing-prototypes]
 1120 | void update_signed_count(struct rtcnt_device_info *dev_info, int rtcnt_count)
      |      ^~~~~~~~~~~~~~~~~~~
/home/ubuntu/RaspberryPiMouse/src/drivers/rtmouse.c:1156:6: warning: no previous prototype for ‘reset_signed_count’ [-Wmissing-prototypes]
 1156 | void reset_signed_count(struct rtcnt_device_info *dev_info, int rtcnt_count)
      |      ^~~~~~~~~~~~~~~~~~
/home/ubuntu/RaspberryPiMouse/src/drivers/rtmouse.c: In function ‘mcp3204_get_value’:
/home/ubuntu/RaspberryPiMouse/src/drivers/rtmouse.c:1780:18: error: implicit declaration of function ‘spi_busnum_to_master’ [-Werror=implicit-function-declaration]
 1780 |         master = spi_busnum_to_master(mcp3204_info.bus_num);
      |                  ^~~~~~~~~~~~~~~~~~~~
/home/ubuntu/RaspberryPiMouse/src/drivers/rtmouse.c:1780:16: warning: assignment to ‘struct spi_controller *’ from ‘int’ makes pointer from integer without a cast [-Wint-conversion]
 1780 |         master = spi_busnum_to_master(mcp3204_info.bus_num);
      |                ^
/home/ubuntu/RaspberryPiMouse/src/drivers/rtmouse.c: In function ‘mcp3204_init’:
/home/ubuntu/RaspberryPiMouse/src/drivers/rtmouse.c:1842:16: warning: assignment to ‘struct spi_controller *’ from ‘int’ makes pointer from integer without a cast [-Wint-conversion]
 1842 |         master = spi_busnum_to_master(mcp3204_info.bus_num);
      |                ^
/home/ubuntu/RaspberryPiMouse/src/drivers/rtmouse.c: In function ‘mcp3204_exit’:
/home/ubuntu/RaspberryPiMouse/src/drivers/rtmouse.c:1871:16: warning: assignment to ‘struct spi_controller *’ from ‘int’ makes pointer from integer without a cast [-Wint-conversion]
 1871 |         master = spi_busnum_to_master(mcp3204_info.bus_num);
      |                ^
/home/ubuntu/RaspberryPiMouse/src/drivers/rtmouse.c: At top level:
/home/ubuntu/RaspberryPiMouse/src/drivers/rtmouse.c:2138:5: warning: no previous prototype for ‘dev_init_module’ [-Wmissing-prototypes]
 2138 | int dev_init_module(void)
      |     ^~~~~~~~~~~~~~~
/home/ubuntu/RaspberryPiMouse/src/drivers/rtmouse.c:2295:6: warning: no previous prototype for ‘dev_cleanup_module’ [-Wmissing-prototypes]
 2295 | void dev_cleanup_module(void)
      |      ^~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
make[3]: *** [scripts/Makefile.build:243: /home/ubuntu/RaspberryPiMouse/src/drivers/rtmouse.o] Error 1
make[2]: *** [/usr/src/linux-headers-6.8.0-1006-raspi/Makefile:1926: /home/ubuntu/RaspberryPiMouse/src/drivers] Error 2
make[1]: *** [Makefile:240: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.8.0-1006-raspi'
make: *** [Makefile:9: rtmouse.ko] Error 2
```

closed 2024-11-07 by KuraZuzu