rt-net/RaspberryPiMouse #98: RaspberryPi OS (kernel: 6.12.25+rpt-rpi-v8)でビルドに失敗する

KuraZuzu, 2025-05-20:
# 不具合の概要

Raspberry Pi OS (64bit)でビルドするとエラーが発生して失敗します。32bit版は未検証です。

# 実行環境

- Raspberry Pi Mouse V3
  - Raspberry Pi 4B+ (4GB)
- Raspberry Pi OS (64bit)
  - 6.12.25+rpt-rpi-v8
- 対象コミット: https://github.com/rt-net/RaspberryPiMouse/commit/484796ea3441b17fc403ec332819c79fda2d7764

# 再現方法

デバイスドライバのダウンロードとデバイスのセットアップをします。

```bash
$ git clone https://github.com/rt-net/RaspberryPiMouse.git
$ cd RaspberryPiMouse/utils
$ ./set_configs.bash
```

ビルドします。

```bash
$ cd RaspberryPiMouse/utils
$ sudo apt install raspberrypi-kernel-headers build-essential
$ ./build_install.bash # ここでエラーが発生
```

# 期待する動作

ビルドが成功し、各種サンプルプログラムで実機が動作する状態

# ログ・画像

`build_install.bash`実行時のビルドエラーのログ全文は以下の通りです。

```bash
$ ./build_install.bash 
build_install_header_from_apt_raspi4.bash
make -C /usr/src/linux-headers-6.12.25+rpt-rpi-v8 M=/home/ubuntu/RaspberryPiMouse/src/drivers V=0 clean
make[1]: Entering directory '/usr/src/linux-headers-6.12.25+rpt-rpi-v8'
  CLEAN   /home/ubuntu/RaspberryPiMouse/src/drivers
make[1]: Leaving directory '/usr/src/linux-headers-6.12.25+rpt-rpi-v8'
make -C /usr/src/linux-headers-6.12.25+rpt-rpi-v8 M=/home/ubuntu/RaspberryPiMouse/src/drivers V=0 modules
make[1]: Entering directory '/usr/src/linux-headers-6.12.25+rpt-rpi-v8'
  CC [M]  /home/ubuntu/RaspberryPiMouse/src/drivers/rtmouse_main.o
  CC [M]  /home/ubuntu/RaspberryPiMouse/src/drivers/rtmouse_dev.o
/home/ubuntu/RaspberryPiMouse/src/drivers/rtmouse_dev.c: In function ‘mcp3204_get_value’:
/home/ubuntu/RaspberryPiMouse/src/drivers/rtmouse_dev.c:180:14: warning: unused variable ‘str’ [-Wunused-variable]
  180 |         char str[128];
      |              ^~~
  CC [M]  /home/ubuntu/RaspberryPiMouse/src/drivers/rtmouse_spi.o
/home/ubuntu/RaspberryPiMouse/src/drivers/rtmouse_spi.c:151:38: warning: ‘struct spi_master’ declared inside parameter list will not be visible outside of this definition or declaration
  151 | static void spi_remove_device(struct spi_master *master, unsigned int cs)
      |                                      ^~~~~~~~~~
/home/ubuntu/RaspberryPiMouse/src/drivers/rtmouse_spi.c: In function ‘spi_remove_device’:
/home/ubuntu/RaspberryPiMouse/src/drivers/rtmouse_spi.c:156:61: error: invalid use of undefined type ‘struct spi_master’
  156 |         snprintf(str, sizeof(str), "%s.%u", dev_name(&master->dev), cs);
      |                                                             ^~
/home/ubuntu/RaspberryPiMouse/src/drivers/rtmouse_spi.c: At top level:
/home/ubuntu/RaspberryPiMouse/src/drivers/rtmouse_spi.c:151:13: warning: ‘spi_remove_device’ defined but not used [-Wunused-function]
  151 | static void spi_remove_device(struct spi_master *master, unsigned int cs)
      |             ^~~~~~~~~~~~~~~~~
/home/ubuntu/RaspberryPiMouse/src/drivers/rtmouse_spi.c:58:26: warning: ‘mcp3204_driver’ defined but not used [-Wunused-variable]
   58 | static struct spi_driver mcp3204_driver = {
      |                          ^~~~~~~~~~~~~~
make[3]: *** [/usr/src/linux-headers-6.12.25+rpt-common-rpi/scripts/Makefile.build:249: /home/ubuntu/RaspberryPiMouse/src/drivers/rtmouse_spi.o] Error 1
make[2]: *** [/usr/src/linux-headers-6.12.25+rpt-common-rpi/Makefile:1974: /home/ubuntu/RaspberryPiMouse/src/drivers] Error 2
make[1]: *** [/usr/src/linux-headers-6.12.25+rpt-common-rpi/Makefile:236: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.12.25+rpt-rpi-v8'
make: *** [Makefile:10: rtmouse.ko] Error 2

```
# その他

YusukeKato, 2025-05-20:
@KuraZuzu 
下記の環境ではビルドできました。
 
- Raspberry Pi Mouse V3
- Raspberry Pi 4B (8GB)
  - `cat /proc/device-tree/model`: Raspberry Pi 4 Model B Rev 1.5
- Raspberry Pi OS (64bit)
  - `uname -r`: 6.6.62+rpt-rpi-v8

カーネルを6.12.25+rpt-rpi-v8に更新後、同じくビルドでエラーが出ることを確認しました。
```
RaspberryPiMouse/src/drivers/rtmouse_spi.c:156:61: error: invalid use of undefined type ‘struct spi_master’
  156 |         snprintf(str, sizeof(str), "%s.%u", dev_name(&master->dev), cs);
```

カーネル更新に合わせてドライバ側も更新が必要そうです。

KuraZuzu, 2025-05-21:
`rpi-6.12`系では、`spi_master`マクロが削除されたために発生しているようです。`spi_master`は`spi_controller`をラップしているだけなため、該当箇所を`spi_master`に置き換えれば動作すると思われます。